kerpz/ArduinoHondaOBD #1: Run on Nano with 328P ?

mr-sneezy, 2015-08-17:
I'm keen to try this code out on my Honda from 1999.
It looks like the code would run on Arduino Nano with 328P which I already have on hand. 
Would there be some reason that it would not be suitable ?
Regards,
Martin

kerpz, 2015-08-17:
Hi martin,

It should run fine on a 328p chip, in fact Im running on a DIY board with 328 chip and UNO bootloader.

Regards, 

Philip B.

mr-sneezy, 2015-08-18:
Great, I did think there would be no problem, but it's good to get 
confirmation.

I'll put this all together later this week and see how it goes on the 
weekend hopefully.
If you want I'll take some photos and make an Eagle schematic for your 
documentation on Github.

I'd like to add some code for checking error codes, and I'm wondering if 
there is enough CPU time available to run a OLED display in lieu of the 
16x2 Hitachi.

Regards,
Martin

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

kerpz, 2015-08-19:
Hi martin,

That would be a great contribution.

Thanks and more power :)

https://www.avast.com/antivirus

mr-sneezy, 2015-08-22:
Update:
The sketch is running on the Nano fine, 16x2 LCD is working fine.
Added HC-05 module and set to 9600baud.
I now have the HC-05 bluetooth talking to Torque app.
I have 'Bluetooth Mode' indication on the LCD.
I see data on the K line pin on the breadboard.
I am yet to connect to the car because my Ebay cable has not arrived, I'll need to make something temporary to do it. 

Philip, do you have a blog or other site where a discussion can be done  ?

kerpz, 2015-08-22:
Hi Martin,
The discussion was done at pgmfi.org before, but I think the site is down.BTW, I'll post some pictures I grabbed from my phone and my car.
Regards,Philip

mr-sneezy, 2015-08-26:
I see the photos Philip, looks great. Looking forward to getting mine finished soon too. 
BTW in researching the Honda OBD protocol I found a list of dlc command requests somebody has compiled, around 200 of them. Do you have something like that already to work from already ?
I could post it here.

kerpz, 2015-08-26:
Yes, anything that could improve this project would be a great help. I think there's a lot to be done in here. code faults scanning is not yet implemented.

mr-sneezy, 2015-08-27:
I hope this is actually useful, I'm collating and cleaning it all up after a google translation (for my own use anyway). This is not the list I mentioned above, as I don't think that one is very relevant after another look at it today. Will look over it again later.

The info below is headed by a URL to the source before each section, so credit goes to the original poster for documenting the info, not me. 
There is a fragmented description of 32 bytes of DTC (from 0x40h to 0x50h it says in the second section) that may help out, also mention of ECU ID on 0x76h in that second block of info. 
--------------------------------------------------------------------
[about Fault Codes ?]
http://hondamotor.ru/board/index.php?showtopic=98035&st=80&p=1626275&#entry1626275

Total: in the log can fall 32 bytes to describe the error. 
These 32 bytes are the description of the 64 bugs.
 such a response is as follows: 00 02 00 20 ... (Hex View)
numbering errors in Honda starts with 0 (0-error ECU, 1 - error of the first sensor O2, 2 - the second O2 sensor error ...)
 in this example 
00 ------> (lower 4 bits 0, upper 4 bits 0) ---> 0 - error0 (no)  0 - error1 (no) 
02 ------> (lower 4 bits 2, upper 4 bits 0) ---> 2 - error2 (yes) 0  -error3 (no) 
00 ------> (lower 4 Bits 0, upper 4 bits 0) ---> 0 - error4 (no)  0 - error5 (no) 
20 ------> (lower 4 bits 0, upper 4 bits 2) ---> 0 - error6 (no)  2 - error7 (yes) 
...........................
So Makar parse all 32 bytes. (The error codes are the same as blink codes) checked in several logs. And local expo error falls under the reduced treatment. 
 
[about Command values (including DTC?) ?]
http://hondamotor.ru/board/index.php?showtopic=98035&st=120
The amount is considered to be just 256 minus the sum of all the bytes of the request or response, if it turns out is less than 0, then add 256
Enum ValuesOffset: byte
RPM = 0x00, // Turnover (2 bytes)
VSS = 0x02, // speed
Flag1 = 0x08, // 15 0-SW Starter 1-SW AC 3-SW Brake
Flag2 = 0x09, // 16 3- SCS self-diagnosis
Flag3 = 0x0A, // 17 2-VTS Vtec Control
Flag4 = 0x0B, // 18 0-MainRelay 2-O2Heater 5 -Engine
Flag5 = 0x0C, // 19 3-VTEC E 7-Econo
Flag6 = 0x0D, // 20
Flag7 = 0x0E, // 21
Flag8 = 0x0F, // 22
ECT = 0x10, // pace antifreeze 
IAT = 0x11, / / temperature
 MAP = 0x12, // the pressure in the inlet 
PA = 0x13, // atmospheric pressure T
PS = 0x14, // dross valve sensor 
O2 = 0x15, // lambda sensor or lean 
Bat = 0x17, // voltage onboard network 
ALTF = 0x18, // Alternator FR Signal (ALTF) signal control board voltage 
CorrCT = 0x20, // krastkovremennaya fuels correction 
CorrLT = 0x22, // fuels long-term correction 
Inj = 0x24, // ignition (2 bytes) 
InjAdv1 = 0x26, // something with the ignition 
InjAdv2 = 0x27, // something with the ignition 
IAC = 0x29, // The Idle air control valve, or IACV regulates the car's idle based on the coolant temperature. 
Knock = 0x3C, // Knock Sensor 

Errors1 = 0x40, // Pars half a byte, half-byte if not equal to 0, it means
Errors2 = 0x50, // there is such an error code. Error number - offset nibbles displacement starts from 0 
VSS2 = 0x61, // speed
RPM2 = 0x62, // Turnover (2 bytes)
ECT2 = 0x64, // pace antifreeze
IAT2 = 0x65, // temperature
MAP2 = 0x66, // the pressure in the inlet
PA2 = 0x67 // Atmospheric pressure
TPS2 = 0x68, // dross sensor zaslonui
Bat2 = 0x69, // voltage onboard network
O22 = 0x6C, // lambda sensor or lean
CorrCT2 = 0x6D, // krastkovremennaya fuels correction
IAC2 = 0x6E, // The air control valve Idle, or IACV regulates the car's idle based on the coolant temperature.
ECUID = 0x76 // ECU ID

Flags]public enum HOBDFlags: ulong
AllOff = (ulong) 0
Starter = 1 << 0
AirCondition = 1 << 1
PASPressure = 1 << 2
VTECPressure = 1 << 7
Brake = 1 << 3
SCS = 1 << (3 + 8), self-diagnostics // 
VTSControl = 1 << (+16 2)
MainRelay 1 << = (0 + 24)
O2Heater << 1 = (2 + 24)
CEL = 1 << (5 + 24)
Engine 1 << = (5 + 24)
VTEC_E = 1 << (3 + 32)
Econo = 1 << (7 + 32)
ACClutch = 1 << (1 + 24)
O2Heater2 << 1 = (6 + 24)
FAN = 1 << (1 + 32)
------------------------------------------------------------------------
If it's all useless rubbish then I apologize in advance :)

mr-sneezy, 2015-08-27:
I have no idea why I have bold text in the last line, it's not shown while editing....

kerpz, 2015-08-28:
Yup found that already, but will double check maybe i missed something.  ￼

mr-sneezy, 2015-08-28:
These bits were of most interest perhaps.

Errors1 = 0x40, // Pars half a byte, half-byte if not equal to 0, it means
Errors2 = 0x50, // there is such an error code. Error number - offset nibbles displacement starts from 0

Total: in the log can fall 32 bytes to describe the error. 
These 32 bytes are the description of the 64 bugs.
such a response is as follows: 00 02 00 20 ... (Hex View)
numbering errors in Honda starts with 0 (0-error ECU, 1 - error of the first sensor O2, 2 - the second O2 sensor error ...)
in this example 
00 ------> (lower 4 bits 0, upper 4 bits 0) ---> 0 - error0 (no) 0 - error1 (no) 
02 ------> (lower 4 bits 2, upper 4 bits 0) ---> 2 - error2 (yes) 0 -error3 (no) 
00 ------> (lower 4 Bits 0, upper 4 bits 0) ---> 0 - error4 (no) 0 - error5 (no) 
20 ------> (lower 4 bits 0, upper 4 bits 2) ---> 0 - error6 (no) 2 - error7 (yes) 
...........................
So Makar parse all 32 bytes. (The error codes are the same as blink codes) checked in several logs. And local expo error falls under the reduced treatment.

mr-sneezy, 2015-08-28:
I note that the author suggests that the error code match the blink code numbering. Would seem logical.

kerpz, 2016-06-24:
Working, on adding this stuff

mr-sneezy, 2016-06-24:
Cool.
As it happens I've restarted the customisation of your code to add some external sensors (3x DS18B20's), to report through to the Torque app. I've dropped the LCD code as it's going to be hard to get enough data on it for my liking and I don't really want to mount a permanent LCD in a classic car like the S2000.
What may help us both is that I have a bench test ECU running now, with original MAP and ECT plus pots for TPS, IAT, O2. With this I could force on fault codes to test with you.
[image]

kerpz, 2016-06-25:
Wow, that's great :)

closed 2021-05-30 by kerpz