kerpz/ArduinoHondaOBD #2: Success and feedback.

mr-sneezy, 2015-09-12:
Hi Philip. 
I took the Ebay Honda OBDII cable and cut off the 16-pin plug end (mine came wired for CAN on pin-14 not K-line on pin-7 anyway). I connected up the ArduinoNano/LCD/Bluetooth on the protoboard to it, and we had nearly instant success on my 1999 Honda S2000 (Australian delivered version). I get data from the HondaOBD converter on my Android phone through the Torque app, I just had to set the adapter speed to slow in the Torque settings first.

Feedback.
I have not yet fully setup Torque with the relay indications like you have, but the basics were working with the default gauge set. 
The data mostly looks right, but coolant temp indicates around 100C at a real measures 80C (I used a thermocouple on the radiator hose and then confirmed again with an IR thermo gun).
Also the coolant temp reads 155C on Torque when the car is off but the Arduino and BT are on. Would be better perhaps to read 0C when there is no valid ECU data perhaps, if this can be done.
Throttle % reads OK, but seems to start at 5.5% at the lowest (5.5% positive offset). I'm not too deep in Honda ECU OBD knowledge but my experience on Nissan Consult was different (0% at 0 throttle pedal deflection).
The data refresh rate is quite slow on torque, I have not looked at how you set the request timing but it would be nice to update data faster if it's at all possible.
Would you know what the theoretical max baud or request rate of the ECU is ? 

I'll have another look at your Torque screen photos and setup my Android device the same and have another look at it tomorrow.
Now that it's running on the protoboard I'm intending to create an Eagle PCB for it. I'll mount the Nano module, the Bluetooth module and regulator on it. LCD too probably but I don't like the white-on-blue one I have right now and will source another. Also going to see what I can do about some indicator leds on the K-line side as some comms feedback to the user (it's nice to know it's running on the ECU side).
Thanks,
Martin

kerpz, 2015-09-13:
Awesome

About the coolant values the readings comes from th ECT sensor, which is attached on the engine block. and when the car is off I think the reading will be the last data since it will return a NO_DATA value.

mr-sneezy, 2015-09-14:
Philip have you considered putting a warning LED driven from the Arduino, that triggers at preset ECT (say 100C) ?
I am looking for a RGB backlit 16x2 LCD as an option to change colour when values go out of normal range. Adafruit has them, and few other vendors...

kerpz, 2015-09-15:
Yes, I have them in the latest commit, I used it to avoid overheating, the current value is 98 degrees. I used a piezo buzzer at pin 13

closed 2021-05-30 by kerpz