kerpz/ArduinoHondaOBD #3: Question about ECT formula ?

mr-sneezy, 2015-09-26:
Hi Philip, where did you get the formula for the ECT calculation in degrees C ?
Regards,
Martin

kerpz, 2015-10-02:
I think from the hondamotors.ru furom they have the formula

mr-sneezy, 2015-10-04:
I found a look up table apparently read from an OBD1 ECU if I understand it right. here on an old Honda forum.
http://www.getglitched.com/pgmfiwiki/www.pgmfi.org/twiki/bin/view/Library/EngineCoolantTemperature.html
If it's valid then it perhaps could be used as a table in in the HondaOBD code, or used to validate the formula currently used.

I also found this website which can derive the formula for you from points on the curve.
http://rusefi.com/articles/measuring_temperature/Steinhart-Hart.shtml

Here's the sensor curve that Honda quote for sensor testing.
[image]

mr-sneezy, 2015-10-04:
[image]
[image]

kerpz, 2015-10-14:
nice

kerpz, 2015-10-14:
why is the E16 is less than I17 intake is hotter than the engine :D

closed 2021-05-30 by kerpz