kerpz/ArduinoHondaOBD #4: connect with chrome ecu software

dibugreen, 2016-01-20:
hi .. 
i'm blind about arduino :).. wuff ..  see lot of awesome project based n arduino .. including this one :)..
is possible to connect your project with  FT232RL FTDI USB to TTL ..(maybe chrome application will accept this communication..)
sorry for my english..
and thx for advance

kerpz, 2016-01-26:
Yes, you can use arduino to connect your ecu to crome application, eliminating the use of usb ftdi to ttl connector.

dibugreen, 2016-01-29:
thx  very much .. i'll try lean arduino .of course based on your project help me very much  ..thx u very much sir ..

mr-sneezy, 2016-01-30:
Crome PC app connection.
Do you both mean to a PC via the bluetooth link ?
If so what Hardware and Port setting are used in Crome to connect to a stock ECU ?
Regards,
Martin

kerpz, 2016-03-07:
yes I think its possible with bluetooth too.

closed 2021-05-30 by kerpz