kerpz/ArduinoHondaOBD #6: Creation of the PID's supported bytes ??

mr-sneezy, 2016-01-27:
Hi Philip,
I've been trying to reverse the OBDII PID supported bytes you send to Torque, but I'm having trouble making sense of them.
else if (!strcmp(btdata1, "0100")) {
    sprintf_P(btdata2, PSTR("41 00 BE 3E B0 11\r\n>"));

I think BE 3E B0 11 are the bytes sent indicating to Torque which of the next 20h PID numbers are used (valid). I can't make sense of the bits though when converted to 32 bits (flags) compared to what's available from the Honda OBD (converted to OBDII numbering).
Is BE the high byte and 11 the Low ?
Did you create some sort of table to describe these bytes from HOBD to OBDII ?

kerpz, 2016-01-27:
Hi Martin,

I used this as my reference https://en.wikipedia.org/wiki/OBD-II_PIDs and the 0100 is a hardcoded reply which is based from https://en.wikipedia.org/wiki/OBD-II_PIDs#Mode_1_PID_00.

mr-sneezy, 2016-01-28:
Hi Philip,  thanks I used that reference too, I converted the four Hex values to 32 bits of binary as they describe but I can't see it align with the OBDII PIDs used by the code. If you have it can you post the binary strings you used to create the Hex bytes, it will help me know what I'm doing wrong from the Wiki info  (I hope).

mr-sneezy, 2016-01-28:
Also just see that the last bit of the four bytes is a flag to validate the next sequence 32 bits....
:{

mr-sneezy, 2016-01-28:
Also, I think the code you wrote below is not needed as the next flag PID after 20h is at 40h, not 30h

else if (!strcmp(btdata1, "0130")) {
    sprintf_P(btdata2, PSTR("41 30 20 00 00 01\r\n>"));

Excuse me if wrong :)

kerpz, 2016-02-03:
Yes, its a dead code afaik :)

closed 2021-05-30 by kerpz