kerpz/ArduinoHondaOBD #7: Using bit values in Torque app, not working for me ?

mr-sneezy, 2016-02-06:
Hi Philip,
Hey you may be able to help me understand why I can get the Torque app to show active relays and switches like you do, even though the ECU and Arduino are TX'ing to Torque.
 In the 'Test' function of the app's Edit Custom OBD2 PID editor (same as you images but in the edit option) I can see the brake and A/C flag bit changing in the third byte of the returned string shown, but I can't see the Equation output change from 0.0 when using the bit formula you use. 

In your photos you show the method to get the bits from byte B, but should that not be byte C going by what the Torque app wiki says ?

Index number of byte in response denoted by Latin alphabet:
A-1 byte, B-2 byte,.., Z-26 byte,AA-27 byte,AB-28 byte,...BA,BB,...

Torque Pro Operations (For reference only)
abs(): Absolute value.
{X:B}: Verify if the bit B is set within the X byte.
[pid]: Refers to the pid number of a sensor.
val{Long Name}: Refers to another equation using its long name.

mr-sneezy, 2016-02-06:
[image]

A screen shot with A/C active in byte string  but not showing in formula..

kerpz, 2016-02-08:
i think its reverse try {B:6}

mr-sneezy, 2016-02-09:
No, it turns out that Torque Pro (current version) takes the first three bytes of the Extended PID response to be header bytes, the fourth byte is A, the fifth is B, six is C etc. This is not documented any place I could see (the three bytes bit).
So the fix was too pad your 20XX commands with an extra byte, then torque works. 
The equation that works then for Brake for example is {A:3}
For one of my extra temp sensors as an Extended PID it's just A

dibugreen, 2016-02-09:
what  protocol is using when connect to torque.thx

mr-sneezy, 2016-06-18:
Seems to be changed now that I updated the Torque app to the current app version. Back to your byte/bit conversion values and it works on my test ECU (now using a 1995 Accord ECU on my bench).

kerpz, 2016-06-20:
Awesome

closed 2021-05-30 by kerpz