kerpz/ArduinoHondaOBD #8: help with obd

tycs03, 2016-08-27:
hi sir, i was wondering if this code will also work for a arduino nano clone. i tried to compile the file and upload it to the nano and wired an hc-06 bluetooth adapter. the code i used was the hobd_elm. i followed the schematic you provided and wired up the bluetooth module to the arduino nano. i wired the rx d11 and tx to d10, k line to d12. im not really sure if that is correct but it seem logical to me.

kerpz, 2016-08-31:
Maybe you can try the hobd_uni, and you need to pair your hc-06 to your phone. BTW, what's the error?

tycs03, 2016-08-31:
i have tried hobd_uni, also i have paired my hc06 to my phone. torque app says no data did you select the right device. seems to be like arduino is not sending anything. By the way i am using a jdm p30 ecu (obd1) i was wondering if i could just wire the arduino directly inside the ecu ? using cn2 port ? i have checked dlc pins and everyting seems to be connected properly. i am thinking maybe i compiled the sketch incorrectly ? i have not much experience with arduino i just bought one cause i saw this hub of yours.

kerpz, 2016-08-31:
I've tested on p2t odb2 and p30 obd1, try to test with elm terminal app.

mr-sneezy, 2016-08-31:
I've tested on Nano with a P0A and a PCX OBDI ECU with no issues either...
If you are building on a protoboard then add indicator LED's with 1k resistors to the TX and RX to bluetooth module, and also to the ECU DLC pin from the Nano. This will tell you what the Arduino is doing easier.
Also the HC06 module needs to be set to use 9600 baud IIRC, and the activity LED should change from fast flash to slower double flash if the Android phone is connecting.

tycs03, 2016-09-01:
i have set the baud rate for the hc06 at 38400 could this be the issue, i set it this way, because chrome rtp requires this baud rate. il try to set it at 9600 see if it works. no its not on a proboard. i just soldered the wires directly to the arduino nano points as i don't have a proto board.

tycs03, 2016-09-01:
Well its working now. Only problem now is that rpm seems to be off. When i rev the engine the meter goes down to 0rpm?

tycs03, 2016-09-01:
And also the rpm is off. Im not sure about the other sensors like ect temp. And map

kerpz, 2016-09-01:
If you are using hobd_uni, the default proto is obd2, set it to obd1 since you are using p30 ecu. Btw are you using hc06 to tune your ecu? Chrome rtp?

tycs03, 2016-09-01:
How to set it to obd1? I haven't  really started to tune my car. I haven't even had the time to install the afr wideband.

tycs03, 2016-09-01:
Also tps has no signal?  (throttle)

tycs03, 2016-09-01:
I was planning to use the hc06 to data log. Althoug seems like the data stream seems to show some lag. Maybe il just use the ttl serial cable instead

tycs03, 2016-09-01:
I forgot currently i am using a stock jdm p30.  But i also have a chipped p28 ecu that i will be using to tune the engine.

tycs03, 2016-09-01:
Ok everything is working now. I was womdering if there  was some way to get tps to work?

kerpz, 2016-09-02:
All sensors are working fine with me.

tycs03, 2016-09-02:
Im not getting anu reading on throttle? And it shows a question mark. When i try to add a display

tycs03, 2016-09-02:
My bad everything is working. Thank you

tycs03, 2016-09-03:
Sir Philip is it possible to add pid for power steering switch?  I have tried the other pid and it works.  I have been meaning to test the brake switch that goes to the ecu and with i this confirmed it to be working. :-) also i have learned the characteristics of the eld. With the ALT PID this is a very nice diagnostic tool as well. Would be really great if it can also pull trouble codes from the ecu.

kerpz, 2016-09-07:
Yes, you can add it through Manage PIDS in torque's setting

tycs03, 2016-09-07:
Sir i dont really know anything about hex.  How do you get the hex values?

kerpz, 2016-10-27:
Hi haven't decode the power steering switch :(

fifthaxe, 2018-06-08:
@tycs03 did you successfully managed to plug the arduino on the CN2 port of your ECU ? Is the CN2 port and the DLC port the same ?

tycs03, 2018-06-13:
Hi. I wasnt able to do the obd gauge. I sold my car. As dlfar as i know its
cb2 and dlc port is not the same.

tycs03, 2018-06-13:
I was able to make the dlc using stock ecu. I never tried to connect it to
cn2 port. My objective for this was to be able to see data from a stock
ecu. Since obd1 ecu runs modfied code you i can see data from a computer.
Uaing crome

tycs03, 2018-06-13:
Oh by the way i have an obd1 with crome and i used bluetooth hc06 to
connect to tuner view. So i use your code on obd2 ecus. Sorry for the
confusion. I though you were the other guy i asked helped from. I never got
to try his code.

Regards,

Tycs03

akilamarquiel, 2018-06-13:
hi guys. ever wonder of using wifi instead of a bluetooth?

closed 2021-05-30 by kerpz