kerpz/ArduinoHondaOBD #9: torque app not working on hobd_uni

abdullah381, 2017-06-01:
hello 
you did very nice job btw. i appreciate your efforts. i used your sketch for my obd1 honda hobd_elm connects to torque app i need to know that can i get the fuel injection rate in miliseconds?

2nd question is the main question. i used hobd_uni . device display everything on lcd but torque app is nt connecting can you please explain the issue however torque is working fine on hobd_elm. 
i am getting rpm in wrong format for example if rpm is 1300 lcd showing 300. very most digit is not working ..

Waiting for your Replay
Thanks for wonderful inovation

kerpz, 2017-06-03:
Hi,
For Question #1 injector duration in ms is given and displayed on LCD page 2
For Question #2 (what ecu are you using? obd1 or obd2) the dafault mode is obd2 so i think you need to set the mode to obd1
Regards

abdullah381, 2017-06-03:
Thanks alot i am using obd1. And please let me know rpm on lcd showing last
3 digit not first digit like if rpm is 1356 screen is showing 356

Thanks

Ouassimf, 2017-06-03:
Somehow same issue,
Torque not reading. Did a debut of the bt serial and Torque is sending a lot of empty msg and the arduino is answering with NO DATA
Nevertheless the first message ATZ or AT0 go through and arduino answers with  Honda OBD 1

Same issue with lcd, Inccorrect RPM (maybe other parameters as well)

And finally when in DLC serial mode, the code never switch to Bluetooth mode despite incming serial data. 
I had to comment out the dlcSerial mode to get bluetooth .

THank a lot for your great program ;)

abdullah381, 2017-06-03:
kerpz 
i have reviewed the code and there is obd1 selected. as Ouassimf decribe above, he is also facing issue with connecting to torque and same RPM issue. can you check the code for us Please? Thanks for creating obd1 diagnotic. take care

abdullah381, 2017-06-05:
Any update? I am waiting for code update to diagnose fuel mileage.pleaseeee

kerpz, 2017-06-06:
I'll try to pull this code and test

abdullah381, 2017-06-14:
no i am using stock ecu. however code is working fine on lcd . i can see parameters

kerpz, 2017-06-14:
how about the rpm on lcd? btw also check hc-05 settings i am using 9600bps baud rate, maybe you are in 38400bps.

eugjr, 2018-02-26:
Hello, I'm facing the same issue. Torque comunicatting with bluetooth, on hodb_elm.
On hodb_uni, the LCD works with rpm errors, and no bluetooth connection with torque.

So far can't identify the difference between the codes.

kerpz, 2018-03-07:
Yep need to check it, some modifications has cause this issue. I guess :(

ExGeniusF, 2018-07-26:
Hello. In the version "hobd_uni" the connection on the bluetooth does not work. Although if you immediately flash "hobd_elm" the option, then the bluetooth normally works without problems.
HC-05 reset to the default settings and re-configured. Did not help. Tell me, what can I try to make it work?

greekuser, 2018-10-08:
hi, first of all a big THANKS to kerpz for the amazing work on producing this.
i have a '98 civic mb and i was designing a pcb and program to wire a bunch of diy sensors to the engine. yeah, i didnt know that my car had an odb1/dlc port for the same thing. So when i realised that, i ordered an elm module, wired the K-line only to see that it didnt work(of course since DLC uses different codes than OBDII). I was totally  amazed to find this project. i had the needed parts, put everything together, used "hobd_elm.ino",bluetooth and TORQUE lite app, connected to the DLC port and BOOM, ECU connected. every data works fine except RPM. comparing the files you will see that in "hobd_elm.ino" the RPM dlcdata is sent to bluetooth without conversion from OBD1 to OBD2. but in "hobd_uni.ino" is converted using "rpm = (1875000 / (dlcdata[2] * 256 + dlcdata[3] + 1)) * 4;" which is the corrent thing to do.remember TORQUE's built in sensors understands OBD2 protocol. so i changed the rpm code  to 
"int rpm = (1875000 / (dlcdata[2] * 256 + dlcdata[3] + 1)) * 4; 
 sprintf_P(btdata2, PSTR("41 0C %02X %02X\r\n>"), highByte(rpm), lowByte(rpm));"
and now RPM shows fine. i hope this helps other too.
Again, kerpz thank you for your time and work. i dont know where you found the honda obd1 protocol codes. i think my car has revived seeing all the data instantly.

closed 2021-05-30 by kerpz