kerpz/ArduinoHondaOBD #10: please help

akilamarquiel, 2017-07-02:
I cannot run on arduinodroid. error. im newbie on programming. please help sir

akilamarquiel, 2017-07-02:
compilation error. and it says 'pushPinHi' and on line 994 it says void and there is unsigned charcter

akilamarquiel, 2017-07-02:
I want to send a screenshot here but I dont know if its possible to see my problem clearer

kerpz, 2017-07-03:
hi I'm not familiar with arduinodroid, I need to know what the error says.

akilamarquiel, 2017-07-03:
I got a screenshot of what I experience. im sorry im new in programming. I
want to learn more and more from expert like you. thank you for your quick
response.  fyi: Ive tried your other .ino file and It was  compiled without
issue. its just the file hobd.ino that has an error. thank you again sir

kerpz, 2017-07-03:
Hi, you mean the hobd_uni.ino? so what .ino did you use?

akilamarquiel, 2017-07-03:
yes sir. I use the hobd_uni.ino sir. the other file was good. I compile it
without any issue. but with hobd_uni.ino, it has error sir. thank you

kerpz, 2017-07-04:
I change char to int, maybe it will fix the compile issue :)

akilamarquiel, 2017-07-05:
sir yes sir. thank you so much. later I will try. I hope I can make it run
on my car. thanks again sir

closed 2021-05-30 by kerpz