kerpz/ArduinoHondaOBD #12: VIN? Unique ID?

SERBice, 2017-09-22:
Is it posible to obtain an unique id or vin from any car to show in the screen about the vehicle??

kerpz, 2017-09-22:
Its specific for honda

SERBice, 2017-09-22:
I want show vin on the screen for my accord 5th generation; f22b1 with p0a ecu.

Is possible obtain vin or uid from ecu on this years?

SERBice, 2017-09-23:
Is it posible to use mode 9 pid 02? How?

Thanks.

mr-sneezy, 2017-09-23:
You need to know the exact byte location of Hondas ECU or VIN data and then translate that to an OBD-II equivalent code. If we don't know this then it can not be done...

SERBice, 2017-09-23:
This Code (Line 547):

`    //byte h_cmd4[6] = {0x20,0x05,0x76,0x0a,0x5b}; // ecu id`

Gets an unique id? How apply it?

Maybe this?:
`dlcCommand(0x20,0x05,0x76,0x0a,data)) `

kerpz, 2017-09-27:
I had tried that byte location but i cannot see any significant data or ecu id on that location :(

SERBice, 2017-10-01:
How do I expel the  dlcCommand function to search for the vin or ID in the ECU ?, I will look for it

Daggers9, 2017-10-01:
You either need to trial and error, or find the correct offset code. 

A Russian dude Reneman has an offset table I'm trying to get a hold of. Dip on pgfmi threw one up ages ago, there is some more info. Kinda dead thread now. 

Btw, nice stuff kerpz, I'm yet to get a junk ECU to test it tho. :)

mr-sneezy, 2017-10-01:
There is some offset info in the first issue thread.
https://github.com/kerpz/ArduinoHondaOBD/issues/1
If you find any info that is more accurate or better than this it would be good to share it back to us here.
I do also have a docx file I compiled from various sources on my own fork.
Thanks,
Martin

SERBice, 2017-11-04:
I would like you to tell me how to "scan" the ecu byte to byte, if possible, of course.

In this way try to find my VIN, since I already know it would be easy to recognize if it is present or not.

How should I use the dlcCommand function to do this? it's possible?

Daggers9, 2017-11-04:
Hey I've been following this for a bit. In my research, what you are looking for is a set of codes, I forget what other people have called them, that allow you to communicate with the ecu. Look through some hodba forums, and look for work done by a guy called Dip/Didier. The PGFMI forums have the best stuff. 

However I don't think you'll find VIN information on an ecu, as an ecu is just a part of the car, not the chassie it's self. It would be like putting the vin on your shocks or wheels or dash. It just doesn't make sense. A VIN is only relevant to the chassie and nothing else.

Daggers9, 2017-11-04:
Lol, thanks myself. you are looking for offset codes.

Daggers9, 2017-11-04:
Sorry the TLDR of it is, the PGFMI forums, have already covered how to trial and error offset codes.

Daggers9, 2017-11-04:
Apologies for the sporadic replies, I currently away from standard internet service, Ive been searching the forums to find the page which explains the trail and error process, but its tough using only my phone. I know I have saved the relevant pages on my computer back home. I'll will be back home in a week, if you post back here around the 14th, I will send you the links I have bookmarked.

Daggers9, 2017-11-20:
http://forum.pgmfi.org/viewtopic.php?f=57&t=5871&start=75&sid=ef6d52919f8a792ea306ae398b685e4c

closed 2021-05-30 by kerpz