kerpz/ArduinoHondaOBD #14: Arduino Uno HC-06 Bluetooth Module

mikewing94, 2018-03-02:
Hello,

I am trying to get this working for my Honda S-MX 1997 (OBD1).

Will this work with the HC-06 Bluetooth module, as my phone doesn't seem to want to connect to it currently?

Thanks

eugjr, 2018-03-05:
Hello, first of all, make sure your HC-06 board is working properly. You can try some simple Arduino codes to make sure you can pair your BT module and cellphone.
Also make sure the wiring is correct.

mikewing94, 2018-03-10:
Thank you, I know the HC-06 board is working properly as I have used it in other projects previously. I will re-do the wiring tonight so I can be certain it is correct though.

mr-sneezy, 2018-03-10:
I use one of the many Android BT serial terminal apps to test my HC-05 modules.

kerpz, 2018-11-02:
Fixed, added a delay on the bt proc

closed 2018-11-02 by kerpz