kerpz/ArduinoHondaOBD #16: Intent to create Wiki

Daggers9, 2018-03-12:


mikewing94, 2018-03-12:
Sounds great! 
Especially for less well known Hondas, (I am currently trying to get this working with a Honda S-MX 1997.)

I think it would also be good if we could get together peoples photographs of wiring etc... I am 99% sure I have a wiring problem with my project currently, but cannot see where from looking at the diagram on here...

Daggers9, 2018-03-12:
My understanding is that this information will be mostly universal for all Hondas that have ecus before 2002.

My project wiring is completely different from this project as I've created an embedded, stand alone device using completely different display components and a bluetooth delete

mikewing94, 2018-03-12:
I'd be happy to get anything working currently, I have had bluetooth issues in the past so a stand alone setup would be fine for me also. Would you mind sharing your setup if possible?

closed 2021-05-30 by kerpz