kerpz/ArduinoHondaOBD #18: Honda 12v to Arduino VIN and Voltage Divider

bonaminGR, 2019-01-01:
Hello !!! I've been trying to put together a diagnostic and datalogging board for my 98 EK, but I stumbled upon a problem. I'm quite new to both programming and electronics so here are my questions ! 

The diagram shows the 12V + Ground pins from my Honda's 3Pin Connector going straight to the Arduino board. But I tried that with an Arduino Pro Mini, and apparently I fried it. :D 

1. Is it possible that I wasn't supposed to connect it like that..?  Do I need to use something BEFORE the board to drop the voltage to smaller numbers...?

2. I used an Arduino Pro Mini and I connected the 12V of the car on the RAW pin. That is the pin where you're supposed to connect up to 12V as I found on the documentation. But when my car is running, isn't it's voltage like 14V or something? Was that the cause ? Or I should not use a pro mini board at all ? 

3. Can I power the board externally using USB or something, and ONLY use the K-Line pin from the car..? 
Or it NEEDS the car's power and ground wires to work ? 

4. There is a mention of a VOLTAGE DIVIDER inside the readme. With two resistor Values. Where each resistor goes and why do I use a voltage divider in the analog0 input on the Arduino ? 

I need some help here folks, I can't figure it out !!
I ordered some new Arduino Pro Mini boards because I don't want to risk the Uno that I'm using as a test bench on my PC. But I need to know if I'm doing something wrong, and how the exact connection should be !

THANKS A LOT IN ADVANCE !!!!
PS. Your project is REALLY interesting ! Especially for a Honda fan like me ! <3

mitchschaft, 2019-01-02:
You need a multimeter to test voltages. But it's supposed to be 12v on the middle pin of the DLC. Are you sure you are using the right connector? How do you know it's fried? Because the arduino pro mini has an onboard input vintage regulator that can handle 16v. And the Uno can handle up to 20v I think. 

What country are you in? I'm not sure if you have both the 3pin DLC and the full 16pin obd2 port. You could use either one.

But it sounds like you connected it right because the mini accepts 12v to raw according to the pinout charts.

mr-sneezy, 2019-01-02:
Some clone Arduinos boards use a cheap regulator that is actually only capable of +12V maximum, and some will blow up around 14V. I've experienced it myself. 
You could put two or three 1N4001 diodes in series on the +12V-RAW line to drop about a volt or so. 
If possible I try to use a quality regulator from car power down to +5V and power the Arduino board from that.
One point, if you have a chipped P30 ECU, does the new chip still support HOBD protocol ?
This might be an unknown...

bonaminGR, 2019-01-02:
> Some clone Arduinos boards use a cheap regulator that is actually only capable of +12V maximum, and some will blow up around 14V. I've experienced it myself.
> You could put two or three 1N4001 diodes [...]

First of all, it's funny how I've been JUST reading this: https://github.com/mr-sneezy/ArduinoHondaOBD_Extra which apparently is your contribution to this Arduino-Honda world ! :D 

I've just contacted my Tuner asking these questions. And apparently my ECU is tuned using Crome Gold. 
Do you know if that actually supports the HOBD protocol ? 

He told me that you can buy datalogging apps, software and hardware to use with ECTune or Hondata S300. But I don't want to do that. I want to experiment. That's the beauty of it.

mitchschaft, 2019-01-02:
What are you guys referring to when you say HODB protocol? Is that another way of saying ODB1? Sorry, I'm from the US and I haven't heard that nomenclature before.
That chipped p30 ecu is still obd1 all the way. Do you know if it's JDM? I guess not if it came from Germany. I believe crome gold is based off the p30 map to begin with.

Otherwise, Crome Gold uses the QD3 protocol to rtp for datalogging. You'll have to have a Crome dealer's license and a usb data cable to get into all that (or cracked Crome software). Or go with another brand like your tuner suggested. A few hundred bucks either way.

Do you know what version of Crome is on your p30? Do you mess with that at all or just take it to him when you need work done? 

As far as your busted arduino minis, reflash the bootloaders using your UNO as an isp and that will get them back up and running.

mr-sneezy, 2019-01-02:
"What are you guys referring to when you say HODB protocol? Is that another way of saying ODB1? "

I'm not sure I've ever seen a document that shows a list of parameters for OBD1, only OBD-II so not sure if  they mean the same thing with Hondas ?
Send me a link to an OBD1 proctol document and I'll compare, or you can have a look at Kerps code and let us know.

I derived the bulk of what I know of the 'HOBD' byte structure and parameters from Kerps code and a bit of Russian software called HOPE (Honda OBD Protocol Examiner).

mitchschaft, 2019-01-02:
Oh no, I think I get it. I'm way out of my depth in that case! I would have a lot of studying to do to catch up on that level and I don't want to waste anyone's time. I have nothing to contribute unless you need anything from a US obd2 96 Accord LX manual f22b2 ecu. 

 I also have an auto ecu lying around from a 97 Odyssey van f22b6 engine I tried to get working with my setup to take advantage of the fuel maps and cam timings after swapping the cam gear from reading on a forum. The ecu seal turned out to be conjecture because you get nothing but a lot of backfire on deceleration even after converting the ecu to manual, which you can do with obd2 just not 100%. It still sees auto components that I think is in the programming. It doesn't do proper fuel cutoff on deceleration. But enough blabbing. I'll get back to building this device.

bonaminGR, 2019-01-02:
My ECU is JDM. I did an engine swap and rebuild, so we used a JDM P30 ECU. 
I found out online, that Crome actually SUPPORTS BOTH the Honda STOCK protocol for Datalogging and its own custom protocol but you must choose which when you're burning the ROM. I will have to check my ECU to see if this is actually the case with me. 
I'm gonna try running the firmware on my Uno but with external power for safety to see what happens and I'll keep you informed if it works with Crome. 

Just for the fun of it, I'm gonna try it on my friends completely stock D14 civic.

bizi27, 2019-04-18:
Hey i have the same problem no info displayed on lcd. It shows the startup but no info about rpm, vss.... All on 0 :/ i tried through js pins from ecu but start to click in ecu so idk what should be wrong

closed 2021-05-30 by kerpz