kerpz/ArduinoHondaOBD #20: I'm trying but nothing

Gnoce, 2019-03-25:
Hi, I'm trying to make this project. I followed the wiring diagram (except for the 12V power supply that I diverted to a stabilizer with 5V output connected to the Arduino 5V). The program has been loaded without errors but the LCD backlights is off and only the squares of the high line are highlighted, both when it's connected to the PC via USB and when it's connected to the 3-pin DLC of the car. The car is a '97 Prelude (P5M). When I connect the DLC, the green Arduino LED and the yellow Rx one light up. If I press the button the Rx turns off and it doesn't turn on again. I had to reset Arduino and reload the program. What can I do? Is there any correlation with the fact that I'm feeding at 5V? I'm using the Uni version.

mr-sneezy, 2019-03-26:
High line dark only means the LCD is not initialized by the MCU properly. Check you have the data and control lines in the right order first as this should work without the car connected (just power).

Gnoce, 2019-03-26:
I will try to connect via IC2, maybe it helps and surely reduces the wiring.

Ottieni Outlook per Android<https://aka.ms/ghei36>

Gnoce, 2019-03-26:
Display is working!!! I've to change 0x3f with 0x27. Now I can try if the DLC work too!

Gnoce, 2019-03-30:
Ok now there is another issue. Everything seems work but after a while the display starts showing strange characters. The only way to get it working again is to turn off the power for at least 15 seconds. The problem occurs randomly after few seconds or minutes, both connected to the car or connected to the PC via USB. If you go past the third minute of this video you can see what happens. Do you have any suggestions? [https://youtu.be/m_F93ymSCmA](url)

mr-sneezy, 2019-03-31:
My experience with getting character corruption on hitachi style LCD's is it's mostly caused by writing to it too fast, or from noise or bad connections on the LCD control lines, or by wires that are too long (having coupling between them), especially bad in the 4-bit data modes.
BTW I have no clue about your post about 0x27, please explain ?

Gnoce, 2019-03-31:
Is the LCD inizialization line "LiquidCrystal_I2C lcd(0x3f, 2, 1, 0, 4, 5, 6, 7); I read that most LCD need another inzialization with 0x27 and not 0x3f. I'm not sure why everyone use this line "LiquidCrystal_I2C lcd(0x27, 2, 1, 0, 4, 5, 6, 7, 3, POSITIVE)" but the hobd_uni code trunk the line at 7. What does "3, POSITIVE" do?

kerpz, 2019-04-10:
Hi Gnoce,

That 2 parameter is for backlight, 3 is the backlight pin and POSITIVE is the polarity.
You can refer to LiquidCrystal_I2C.h

Regards

Gnoce, 2019-04-20:
Ok, I have redone the welds between IC2 and the LCD and now the display seems to work correctly. The problems are now the rev data on LCD and Torque. Torque cannot read also other data, the voltage display random numbers.  hobd_uni setted on OBD1 and HC-05 on 9600 baud . I have to try flashin hobd_elm and see. I forget to thank you for your great job kerpz!!!

closed 2021-05-30 by kerpz