kerpz/ArduinoHondaOBD #21: How to read error with hobd_elm.

Liberator24, 2019-03-31:
Hey.
Is there some way to read throught troque app errors with hobd_elm code or maybe  hobd_uni2 can do this?
Kerpz thank you for your time and work, excelent project.

Liberator24, 2019-04-07:
Is it correct? 
else if (!strcmp(btdata1, "2040")) { // ERROR
    if (dlcCommand(0x20, 0x05, 0x40, 0x10, dlcdata)) {
      sprintf_P(btdata2, PSTR("60 40 %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));

kerpz, 2019-04-10:
Hi Liberator24,

No support for the ELM yet, on the LCD I think page 4 shows the registered error, I haven't verified yet.

Liberator24, 2019-04-10:
With this code i can read most of codes, but only in custom pid testing on troque pro. I do not know how to  pull out this codes just by klicking (scan erorrs) in troque app.
 }
  else if (!strcmp(btdata1, "2040")) { // ERROR
    if (dlcCommand(0x20, 0x05, 0x40, 0x10, dlcdata)) {
      sprintf_P(btdata2, PSTR("20 40 %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X\r\n>"), dlcdata[0], dlcdata[1], dlcdata[2], dlcdata[3], dlcdata[4], dlcdata[5], dlcdata[6], dlcdata[7], dlcdata[8], dlcdata[9], dlcdata[10], dlcdata[11], dlcdata[12], dlcdata[13], dlcdata[14], dlcdata[15], dlcdata[16], dlcdata[17]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }
   }
  else if (!strcmp(btdata1, "2050")) { // ERROR2
    if (dlcCommand(0x20, 0x05, 0x50, 0x10, dlcdata)) {
      sprintf_P(btdata2, PSTR("20 50 %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X %02X\r\n>"), dlcdata[0], dlcdata[1], dlcdata[2], dlcdata[3], dlcdata[4], dlcdata[5], dlcdata[6], dlcdata[7], dlcdata[8], dlcdata[9], dlcdata[10], dlcdata[11], dlcdata[12], dlcdata[13], dlcdata[14], dlcdata[15], dlcdata[16], dlcdata[17]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }

kerpz, 2019-04-11:
Yes it takes time to translate although its doable, you need to see the obd2 pids wiki if you plan to do it.

closed 2021-05-30 by kerpz