kerpz/ArduinoHondaOBD #27: RPM runs down instead rev up

rubensx, 2021-03-22:
Hi Kerps, nice work, I'm using a arduino uno and a HC-06 bluetooth as a simple bridge to torque, but, the RPM goes to another direction instead rev up when I hit the accel pedal!

I'm trying other codes but still no lucky

The ar is a P28 ECU honda EJ1 1995 d16z6 

` else if (!strcmp(btdata1, "010C")) { // rpm
    if (dlcCommand(0x20, 0x05, 0x00, 0x02, dlcdata)) {   //Note 2 bytes used here
        if ((dlcdata[2] == 255) && (dlcdata[3] == 255)) {   //Trap a condition where my test ECU sends 255, 255 when engine is off
            dlcdata[2]=0;                                   //Set a 255 value to 0
            dlcdata[3]=0;
        }
        sprintf_P(btdata2, PSTR("41 0C %02X %02X\r\n>"), dlcdata[2], dlcdata[3]);
    }
    else {
      sprintf_P(btdata2, PSTR("DATA ERROR\r\n>"));
    }
  }`

tested the original code, and this above, and still down the RPM

kerpz, 2021-03-26:
Hi Rubens, you need to change the mode to OBD1

kerpz, 2021-03-26:
`byte obd_select = 1;`  1 for obd1, and 2 for obd2 they have different encodings for rpm, p28 is obd2 I guess, so the value would be 2.

> Hi Rubens, you need to change the mode to OBD1

rubensx, 2021-03-26:
hi Philip, thank you for your help, but I guess I don't know how I put this "byte pbd_select = 1"

rubensx, 2021-03-26:
the P28 ECU is OBD1

kerpz, 2021-03-26:
> the P28 ECU is OBD1

which code did you use? the hobd_uni2?

rubensx, 2021-03-26:
I'm using this hobd_elm.ino and edit some parts, all other codes don't link ecu for me

rubensx, 2021-03-26:
`/*
 Author:
 - Philip Bordado (kerpz@yahoo.com)
 Hardware:
 - HC-05 Bluetooth module at pin 10 (Rx) pin 11 (Tx) 
 - DLC(K-line) at pin 12
 Software:
 - Arduino 1.0.5
 - SoftwareSerialWithHalfDuplex
   https://github.com/nickstedman/SoftwareSerialWithHalfDuplex
   
 Formula:
 - IMAP = RPM * MAP / IAT / 2
 - MAF = (IMAP/60)*(VE/100)*(Eng Disp)*(MMA)/(R)
   Where: VE = 80% (Volumetric Efficiency), R = 8.314 J/°K/mole, MMA = 28.97 g/mole (Molecular mass of air)
   http://www.lightner.net/obd2guru/IMAP_AFcalc.html
   http://www.installuniversity.com/install_university/installu_articles/volumetric_efficiency/ve_computation_9.012000.htm
*/
#include <SoftwareSerialWithHalfDuplex.h>

SoftwareSerialWithHalfDuplex btSerial(10, 11); // RX, TX
SoftwareSerialWithHalfDuplex dlcSerial(12, 12, false, false);

const byte myTPSlow = 14;    //your cars zero throttle TPS byte value (trim to get 0% on OBD2 or leave at 0)
const byte myTPShigh = 231;  //your cars full throttle TPS byte value (trim to get 100% on OBD2 or leave at 255)
byte obd_select = 1;

bool elm_memory = false;
bool elm_echo = false;
bool elm_space = false;
bool elm_linefeed = false;
bool elm_header = false;
int  elm_protocol = 0; // auto

void bt_write(char *str) {
  while (*str != '\0')
    btSerial.write(*str++);
}

void dlcInit() {
  dlcSerial.write(0x68);
  dlcSerial.write(0x6a);
  dlcSerial.write(0xf5);
  dlcSerial.write(0xaf);
  dlcSerial.write(0xbf);
  dlcSerial.write(0xb3);
  dlcSerial.write(0xb2);
  dlcSerial.write(0xc1);
  dlcSerial.write(0xdb);
  dlcSerial.write(0xb3);
  dlcSerial.write(0xe9);
  delay(300);
}

int dlcCommand(byte cmd, byte num, byte loc, byte len, byte data[]) {
  byte crc = (0xFF - (cmd + num + loc + len - 0x01)); // checksum FF - (cmd + num + loc + len - 0x01)

  unsigned long timeOut = millis() + 250; // timeout @ 250 ms
  memset(data, 0, sizeof(data));

  dlcSerial.listen();

  dlcSerial.write(cmd);  // header/cmd read memory ??
  dlcSerial.write(num);  // num of bytes to send
  dlcSerial.write(loc);  // address
  dlcSerial.write(len);  // num of bytes to read
  dlcSerial.write(crc);  // checksum
  
  int i = 0;
  while (i < (len+3) && millis() < timeOut) {
[... 389 lines cut ...]

  
  else {
    sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
  }

  if (btdata2) bt_write(btdata2); // send reply
}  

void setup()
{
  //Serial.begin(9600);
  btSerial.begin(9600);
  dlcSerial.begin(9600);

  delay(100);
  dlcInit();
}

void loop() {
  btSerial.listen();
  if (btSerial.available()) {
    procbtSerial();
  }
}`

rubensx, 2021-04-01:
hi Phillip, I'm stuck here =( can't solve the RPM

kerpz, 2021-04-01:
> hi Phillip, I'm stuck here =( can't solve the RPM

hi rubensx, found the issue hobd_elm use the obd2 algo

`  else if (!strcmp(btdata1, "010C")) { // rpm
    if (dlcCommand(0x20, 0x05, 0x00, 0x02, dlcdata)) {
      //sprintf_P(btdata2, PSTR("41 0C %02X %02X\r\n>"), dlcdata[2], dlcdata[3]);
     int rpm = (1875000 / (dlcdata[2] * 256 + dlcdata[3] + 1)) * 4;
     if (rpm < 0) { rpm = 0; }
     sprintf_P(btdata2, PSTR("41 0C %02X %02X\r\n>"), highByte(rpm), lowByte(rpm)); //((A*256)+B)/4
    }
    else {
      sprintf_P(btdata2, PSTR("DATA ERROR\r\n>"));
    }
  }`

you need to replace that with

rubensx, 2021-04-03:
hi Philip!! awesome!! works great!!! but  when engine is off the rpm goes to 17000 he he he, but solved on engine running... hey you have see this ? (hondash) ? has some features more (and I guess is the same project here)

closed 2021-05-30 by kerpz