kerpz/ArduinoHondaOBD #28: help with translate honda code ECT to a number

rubensx, 2021-04-23:
Hi Philip! how are you? 

I came here because I had no one who know too much about honda code and maybe you can help
I'm in a project (a TFT ECT ring meter) to solve a S2000 AP2 2007+ problem (this cluster has a rocket science method to see ECT and need a CAN bus decode... yep forget this CAN and I'm creating a external temperature gauge).

Ok, using a ESP32 and a FTF 3.2" screen we can get a nice ring meter to place over the AP2 (with the same design) but I need to get a ECT value as a decimal value... 

Looking your code I guess can use only ECT data to do that... (maybe you ask me about use a termopar k sensor or read the ECT gauge sensor from D16Z6, but the wire length and the engine temperature and RF interference is so bad to a simple arduino. So, I'm thinking about convert from 3 pin DLC information directly to my gauge (maybe a VTEC information too).

Can you take me to a "north" about this? 

Thank you again (if you want to see this project I can show you the ring code images about)

rubensx, 2021-04-23:
[image]

Maybe a strange solution but...

mr-sneezy, 2021-04-24:
I'm not Philip, but I do have half of his original code in my branch of his project, on an S2K AP1.

If the 2007+ S2K is CAN bus then I'd buy a CAN transceiver for the ESP32 and load up somebodies CAN sniffer code. It may not be that hard to workout the packet structure etc. I'm surprised there is not already a CAN project for Honda on Github.

On the other hand, if all you want is ECT and are going to splice onto the gauge sensor wire and read the voltage into the ESP32 ADC, it should work. You may need to run into a 1:2 voltage divider to not exceed 3.3V on the ADC input. I have found from experience you will need at least a 10uF filter capacitor on the ADC to get rid of EMI and other electrical noise (like with my own additional S2K oil pressure sensing). 

If you want a sensor that you can glue to the engine or coolant hose that WILL work over the length of the car and be immune to noise go with the DS18B20 type. I have three of these on a single 1-wire bus on my S2K and it behaves perfectly. One goes right back to the differential.

rubensx, 2021-04-24:
Hi Sneeze!! Your work is awesome! 

Yea I’m trying to use CAN bus but some problems:
1- my version of AP2 is 2007+ and the ECT are from a delta by oil temp - ambient temp and coolant... is much complex and need a entire S2000 to sniff CAN bus (I’m at Brazil and S2000 are extremely rare and his parts too)
2- I bought a used cluster and someone edit parts of this cluster and maybe affect the CAN transceiver and since the cluster are a receptor and not a sender os these messages, it’s impossible to discover the right messages
3- at last, about all this situation, using a ILI9431 TFT on a ESP32 (yes in my first tests using this board to sniff CAN and other tests), and create a bezel to put over the panel.

this idea not so much better but I think some future uses in this TFT like a VTEC light, turbo booster information ando some other info by a push button... but I need to up the stairs by the first stage! Need a ECT... in my desk using a IR thermo and mlx library working good... I’ll see this DS18B20 but if I use the information from ecu is more accurate and maybe use more ECU information from my auxiliary’s TFT in future
[image]
[image]
[image]
[image]
[image]
[image]
[image]

rubensx, 2021-04-24:
[image]
[image]
[image]
[image]

rubensx, 2021-04-24:
Is not finished and the TFT not centered and has a plastic protection but... the animation and react is perfect and in real world are better than the photos

mr-sneezy, 2021-04-25:
Nice work so far, what you need is an ECU and a dash loom from a wreck to make a test system.

rubensx, 2021-04-25:
Import to Brazil is too expensive and the cost don’t worth! In this case a nice way is a Fueltech FT550 with touchscreen in dashboard!

if you can help to decode dlc ECT to a decimal number is desirable

rubensx, 2021-04-25:
I'm using the OBD_UNO version and works great to see in android, but I need a function to enable ECT codes to send without a BT connection (send via serial).

`  else if (!strcmp(btdata1, "0105")) { // ect (°C)
    if (dlcCommand(0x20, 0x05, 0x10, 0x01, dlcdata)) {
      float f = dlcdata[2];
      f = 155.04149 - f * 3.0414878 + pow(f, 2) * 0.03952185 - pow(f, 3) * 0.00029383913 + pow(f, 4) * 0.0000010792568 - pow(f, 5) * 0.0000000015618437;
      dlcdata[2] = round(f) + 40; // A-40
      sprintf_P(btdata2, PSTR("41 05 %02X\r\n>"), dlcdata[2]);
      
      // Test monitor
      ECT = f; // ECT declared as a float above
      Serial.print(F("*ECT "));
      Serial.println(ECT);
     
    }
    else {
      sprintf_P(btdata2, PSTR("DATA ERROR\r\n>"));
    }`

But only can see the ECT decimal with bluetooth android logged

any idea to start sending this value to serial bt independent?

rubensx, 2021-04-26:
stuck here... 

for a strange reason only the ELM_Bridge  works for me to conect my bluetooh android cell 
My goals is:

- Create a dlc bridge who can take all informadion by the pids and send it for serial (a decimal numbers)
- with ESP32 i'll take this decimal numbers to show ECT and VTEC light
- if I can still use a bt ELM connection is awesome

At the moment, I only can send serial decimal numbers message when the cell BT is connected, and I need it independent...

some ideia about it?

marcus-1975, 2021-04-27:
Hi, I don't have time to be doing detailed coding descriptions as I'm busy with my own projects, but in short here is were I'd start. Start with cloning Philips HOBD-UNI code version, disable the calls to the bluetooth code by disabling checks for BT data available (so it then uses LCD code block), then add a serial.print command right after the ECT calculations, to output the ECT values each cycle to the UART (that you choose to setup). Crude but should work. The Arduino does know you don't have the LCD so it can bang away on those pins care free. 
https://www.arduino.cc/reference/en/language/functions/communication/serial/print/

Good luck with the project, you will learn a lot.

mr-sneezy, 2021-04-27:
Hi, I don't have time to be doing detailed coding descriptions as I'm busy with my own projects, but in short here is were I'd start. Start with cloning Philips HOBD-UNI code version, disable the calls to the bluetooth code by disabling checks for BT data available (so it then uses LCD code block), then add a serial.print command right after the ECT calculations, to output the ECT values each cycle to the UART (that you choose to setup). Crude but should work. The Arduino does know you don't have the LCD so it can bang away on those pins care free.
https://www.arduino.cc/reference/en/language/functions/communication/serial/print/

Good luck with the project, you will learn a lot.

rubensx, 2021-04-27:
Thank you Martin! Yesterday I’m using this solution (LCD version)!

In fact I’m trying to maintain Bluetooth connection to have a 2 options (a ECT for the s2000 panel and if I need a troubleshooting, using a torque app) but is more complex and I’ll test!

For some strange reason no other code from you and Philip work in my car a civic ej1!

when I open a torque app, this never “found a protocol” only a obd_elm bridge or obd_uno works...

the protocol handshake betwen other versions seems different (maybe the choose of obd1/obd2) and don’t work at all...

thank you for the north and I’m still trying something to include all functions and at last I will post the result

cheers!!

kerpz, 2021-04-30:
Hi guys, awesome work you got there. I've been busy with my new work so I haven't had time to tinker with this project. If you don't want to decode or read CAN bus I think you can read the sensor signal directly. I've done it before and derived the code from mpguino, so it's a standalone system reading speedometer, ect, rpm, injectors, etc. I'm not sure on where or what hard drive did I stored the codes. :)

rubensx, 2021-04-30:
Hi Philip!! How are you! Wow new work is exciting!! Go ahed!!

if you find rhis work and have a ECT values makes my life much more easy! If you can at this time! If this code are in internet maybe I can find in google maybe?

thank you

closed 2021-05-30 by kerpz