kerpz/ArduinoHondaOBD #30: Throttle inverse on obd1

rubensx, 2022-09-05:
hi kerps!

I can't figure out about my Throttle appers inverse (from 100% to 0) in bluetooth mode linked into Torque app. At the LCD all data are ok!

Can you help?

is a P28 ECU d16z6

thank you

kerpz, 2022-09-06:
Hi Ruben, try to set the obd mode to obd1.

rubensx, 2022-09-06:
I'll try, but P28 ECU are OBD1 the handshake will match?

rubensx, 2022-09-06:
[https://cdn2.solojavirtual.com/loja/arquivos_loja/59350/Fotos/thumbs3/produto_Foto2_12176017.png?cache=](url)

I'm using a arduino mega pro and a i2c oled to put all in clock set into a EJ1 (coupe 1995). I have do some changes into code for fit in this project!

rubensx, 2022-09-07:
hi Kerps!

I'm back! 
I do a mistake! it's a R throttle dial in torque is show inverse! But I can't set to obd2 intead obd1!

but I test other files like the obd_elm and others and another doubt:

```
   else if (!strcmp(btdata1, "0145")) { // iacv / relative throttle position
    if (dlcCommand(0x20, 0x05, 0x28, 0x01, dlcdata)) {
      sprintf_P(btdata2, PSTR("41 45 %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("DATA ERROR\r\n>"));
    }
  }
  else if (!strcmp(btdata1, "2008")) { // custom hobd mapping / flags
    if (dlcCommand(0x20, 0x05, 0x08, 0x01, dlcdata)) {
      sprintf_P(btdata2, PSTR("60 08 %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }
  }
  else if (!strcmp(btdata1, "200B")) { // custom hobd mapping / flags
    //sprintf_P(btdata2, PSTR("60 0C AA\r\n>")); // 10101010 / test data
    if (dlcCommand(0x20, 0x05, 0x0B, 0x01, dlcdata)) {
      sprintf_P(btdata2, PSTR("60 0B %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }
  }
  else if (!strcmp(btdata1, "200C")) { // custom hobd mapping / flags
    if (dlcCommand(0x20, 0x05, 0x0C, 0x01, dlcdata)) {
      sprintf_P(btdata2, PSTR("60 0C %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }
  }
  else if (!strcmp(btdata1, "200F")) { // custom hobd mapping / flags
    if (dlcCommand(0x20, 0x05, 0x0F, 0x01, dlcdata)) {
      sprintf_P(btdata2, PSTR("60 0F %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("NO DATA\r\n>"));
    }
  }
  
 //Custom PID for IACV position (Idle Air Control Valve opening) From HOPE
  else if (!strcmp(btdata1, "2004")) { // IACV position value (scaled to % by Torque app)
    if (dlcCommand(0x20, 0x05, 0x28, 0x01, dlcdata)) {
    sprintf_P(btdata2, PSTR("60 04 %02X\r\n>"), dlcdata[2]);
    }
    else {
      sprintf_P(btdata2, PSTR("DATA ERROR\r\n>"));
    }
  }
```

at the code above you put a 4 digit 2004 (as a example) to the custom pids, but in new version you create anbother way to identifi with only 2 digits. May this not clear to me ho you change the pid 0145 to the 2004 identify in obd_uni2 file?

```
          }
          else if (strstr(&btdata1[2], "42")) { // ecu voltage (V)
            if (dlcCommand(0x20, 0x05, 0x17, 0x01)) {
              float f = dlcdata[2];
              f = f / 10.45;
              unsigned int u = f * 1000; // ((A*256)+B)/1000
              sprintf_P(btdata2, PSTR("41 42 %02X %02X\r\n>"), highByte(u), lowByte(u));
            }
          }
          else if (strstr(&btdata1[2], "45")) { // iacv / relative throttle position (%)
            if (dlcCommand(0x20, 0x05, 0x28, 0x01)) {
              sprintf_P(btdata2, PSTR("41 45 %02X\r\n>"), dlcdata[2]);
            }
          }
        }
```

like this? Because I don't get clue how you know if the first 2 digits are 01 or 20!

That's I say because the obd_elm works great in P28 ecu with custom pids, but with new version some information are missing!

thank you again for your assistance and work

rubensx, 2022-09-07:
more tests: 

with your custom PIDs file in torque folder, all changes runs like charm! Maybe my problem is, in LCD TPS = "-6" and in torque app the TPS are 98%, when I hit the gas slowly, became to 100% and runs to 0, 1, 2 ,3 ,4% (like the scale are looping and don't show negative scale in torque... weird)

rubensx, 2022-09-07:
How I can contribute with you to test or help?

kerpz, 2022-09-13:
Hi @rubensx sorry been busy lately, thanks for bringing that up I will check and compare it with the code that I'm using.

kerpz, 2022-09-19:
Hi @rubensx fixed applied to hobd_uni2, corrected the PID mapping for relative tps and iacv. I cannot test it at the moment my car is currently being restored.

rubensx, 2022-09-20:
Great! I’ll test for you in my “scanner” and I’ll back with feedback!

rubensx, 2022-09-21:
Hi @kerpz I'm using your sketch without any extra sensors, much like "hondash" and scanner only. So, in my tests with hc06 bt and arduino uno (buzzer and button present). So I need to edit some data, but now with last sketch in obd1 the bluetooth connect but TORQUE never found the handshake to ecu.

mix the code using only your TPS and IACV edit to old code, bt connect and running but tps and iacv are dead

kerpz, 2022-09-22:
Hi @rubensx  the hc06 has different baudrate 38400 if I remember it correctly.

kerpz, 2022-09-22:
Hi @rubensx  what code are you using? hc06 has 38400 baudrate by default.

rubensx, 2022-09-22:
My Hc06 works on 9600 (don't work in 38400),

Your newer code don't handshake the ecu protocol, and my code is slightly modified to exclude all extra sensors only has a LCD and a button

```
/*
 Author:
 - Philip Bordado (kerpz@yahoo.com)

 Hardware:
 - Arduino Nano (Compatible board)
 - HC-05 Bluetooth module
 - Resistors 680k ohms and 220k ohms
 - Piezo Buzzer
 - LCD 16x2 I2C
 Optional:
 - 100 psi transducer for fuel pressure (0.5v - 4.5v)
 - AEM AFR UEGO
 - 10K NTC Thermistor

 Software:
 - Arduino 1.8.19 (Stable)
 - SoftwareSerialWithHalfDuplex (Library)
   https://github.com/nickstedman/SoftwareSerialWithHalfDuplex

 Formula:
 - IMAP = RPM * MAP / IAT / 2
 - MAF = (IMAP/60)*(VE/100)*(Eng Disp)*(MMA)/(R)
   Where: VE = 80% (Volumetric Efficiency), R = 8.314 J/°K/mole, MMA = 28.97 g/mole (Molecular mass of air)
   http://www.lightner.net/obd2guru/IMAP_AFcalc.html
   http://www.installuniversity.com/install_university/installu_articles/volumetric_efficiency/ve_computation_9.012000.htm

 Arduino Pin Mapping / Configuration:
 * Pulse Input - 50k ohms over 5.1v zener
 $ Switch Out - 1k ohms on NPN transistor (2n3904)
 
 - 00 = Serial RX
 - 01 = Serial TX
 - 02 = * Injector Input
 - 03 = * Injector Input
 - 04 = $ Door Lock | * RPM Input
 - 05 = $ Door Unlock | * VSS Input
 - 06 = $ AC Compressor
 - 07 = 
 - 08 = * Door Sensor
 - 09 = AEM AFR UEGO (Serial In)
 - 10 = Bluetooth RX
 - 11 = Bluetooth TX
 - 12 = K-Line (Serial In)
 - 13 = Piezo buzzer (+)

 - 14 = (A0) Voltage divider (Input Signal)
 - 15 = (A1) 100 PSI Fuel Pressure
 - 16 = (A2) AEM AFR UEGO (ADC)
 - 17 = (A3) NTC Thermistor
 - 18 = (A4) I2C / LCD I2C
 - 19 = (A5) I2C / LCD I2C
 - 20 = (A6) Navigation Button (reason: multiple button using resistors)
 - 21 = (A7)
*/

#define APPNAME " Honda OBD v1.0" // 16 chars

#include <EEPROM.h>

// OUT
//#define PIN_LOCK 4
//#define PIN_UNLOCK 5
//#define PIN_AC 6
#define PIN_BUZZER 13
[... 1090 lines cut ...]
  
  if (!isButtonPressed) {
    btSerial.listen();
    if (btSerial.available()) {
      if (!elm_mode) {
        elm_mode = true;
        lcd.clear();
        lcd.setCursor(0, 0);
        lcd.print(" Bluetooth Mode");
        lcd.setCursor(0,1);
        lcd.print(" ECU Type: OBD");
        lcd.print(obd_select);
      }
      procbtSerial();
      btTick = millis();
    }
    if (millis() - btTick >= 2000) { // bt timeout 2 secs
      elm_mode = false;
    }
    if (!elm_mode) {
      execEvery(250);
    }
  }
}
```

rubensx, 2022-09-22:
but this code above don't match TPS and IACV or engine load (R Throttle)

rubensx, 2022-09-22:
I'm creating a OLED tiny screen version, change some lcd codes only (to put in clock housing in dashboard)

kerpz, 2022-09-23:
Yup haven't fine tune the DTC scanner but at least it works on the LCD, the handshaking needs a fine tuning as well could be the timing or something. Meanwhile still waiting for my paint job. :)

rubensx, 2022-09-23:
Your work is great, man! I’m thankful for your patience and skills! It’s a great hobby for me in these pandemic years :) and help me to solve various issues in my car (I bought a EJ1 from a junkyard, and restore from ZER0! Almost done some modifications! And your project will be my “board computer in clock”

this code up in last message are my last test but still tps and iacv not workin in torque app but in LCD all are ok! 

Your car is a dc2?

rubensx, 2022-10-03:
still no lucky with tps in torqu app =(

closed 2022-10-10 by rubensx